spacetelescope/poppy #200: afocal systems break focal length and platescale calculations

mperrin, 2018-08-23:
[image] **Issue by [douglase](https://github.com/douglase)**
_Tuesday Jan 03, 2017 at 18:02 GMT_
_Originally opened as https://github.com/mperrin/poppy/issues/200_

----

As discovered recently by @neilzim and as [previously](https://github.com/douglase/poppy_example_notebooks/blob/master/Fresnel/Microscope_Example.ipynb) noted but not documented, the current way plate scales are calculated in the fresnel wavefront class depends on the focal length attribute carried through the sequence of propagations by the wavefront class. Thus, it breaks when you have an afocal system where plane-to-plane propagation occurs. 

We need either another way of carrying focal length through a system, or more graceful failure and better documentation.

isa-kain, 2023-05-24:
Hi @mperrin! I'm running into this bug as well, where the focal length is "forgotten" in collimated space and the plate scale becomes -inf. Has there been any progress on this, or do you have a suggested quick fix?

mperrin, 2023-06-05:
@isa-kain Thanks for your patience & apologies for not being able to write back sooner. I have not had a chance to look at this yet. 

It might help if you could share some example code that runs into this bug, maybe a simple simulation of an focal simulation like this in a script or notebook. Having a runnable code example would help me get started on looking into this. Cheers.