spacetelescope/poppy #253: Implement ThermalBloomingWFE

mperrin, 2018-08-27:
[image] **Issue by [DaPhil](https://github.com/DaPhil)**
_Friday May 18, 2018 at 09:55 GMT_
_Originally opened as https://github.com/mperrin/poppy/issues/253_

----

I implemented the capability to calculate thermal blooming effects with poppy. For this purpose, I wrote a class PhysicalFresnelWavefront (see #248). I wanted to avoid making another pull request because you guys seem to be quite busy. But have a look at my [thermal blooming branch](https://github.com/DaPhil/poppy/tree/thermal_blooming) and as sneak peak at the [notebook](https://github.com/DaPhil/poppy/blob/thermal_blooming/notebooks/Thermal%20Blooming%20Demo.ipynb) too. I haven't implemented any tests yet but let me know if this is interesting for you.

mperrin, 2018-08-27:
[image] **Comment by [mperrin](https://github.com/mperrin)**
_Friday May 18, 2018 at 12:25 GMT_

----

Thanks. Yes, given other commitments and deadlines I probably won’t look at this for a while. Likely not until after the SPIE conference in June.

DaPhil, 2021-07-29:
@mperrin: My pleasure to contribute. I think this issue can now be closed.

mperrin, 2021-07-29:
Added in #438

closed 2021-07-29 by mperrin