spacetelescope/poppy #257: re-importing modules breaks fresnel propagation

mperrin, 2018-08-27:
[image] **Issue by [douglase](https://github.com/douglase)**
_Sunday Jun 03, 2018 at 17:01 GMT_
_Originally opened as https://github.com/mperrin/poppy/issues/257_

----

@mperrin, I found the bit of code that was breaking coronagraph sims during benchmarking. 

This is an edge case, but running the following snippet to reload modules after importing causes optics to not be applied while calc_psf is called. The propagation occurs and wavefronts are multiplied by optics, but without rescaling the wavefront pixelscale, so the optic arrays are interpolated to zero size and no flux is observed at the output of the system. 

Since the lens power is not applied, this can also change the number of FFTs performed, invalidating benchmark numbers.

 This can be diagnosed from the debug statements, where there are no
 "DEBUG:------ Optic: " statements. Otherwise the failure is silent. 

```
import importlib
importlib.reload(poppy.fresnel)
importlib.reload(poppy.optics)
importlib.reload(poppy)
```

Under the new GPU selection paradigm, #250, this hack to set the _USE_CUDA etc.. keywords aren't needed to switch acceleration modes so this shouldn't be problem in practice but is being documented here in case it comes up again.

mperrin, 2018-08-27:
[image] **Comment by [mperrin](https://github.com/mperrin)**
_Sunday Jun 03, 2018 at 18:46 GMT_

----

Nice debugging to track down where that glitch was coming from. Not so surprising- reloading modules is pretty brittle and unreliable.  I’m sure this is related to some calls to `isinstance` which don’t work as expected after reloading.

mperrin, 2018-08-27:
[image] **Comment by [mperrin](https://github.com/mperrin)**
_Monday Jul 09, 2018 at 17:20 GMT_

----

Do we think there is actually any change needed for this, or is it just a matter of 'don't use the reload function, it's not reliable'?

mperrin, 2018-08-27:
[image] **Comment by [douglase](https://github.com/douglase)**
_Monday Jul 09, 2018 at 17:27 GMT_

----

The latter is probably OK for now, do we need to include it in a troubleshooting doc or do we feel this issue is this sufficient documentation?

mperrin, 2018-11-28:
@douglase do you have any objections to my closing this issue? I will consider this closed if I don't hear an objection by the end of this week. I think that's fair since this was related to a test notebook issue rather than any problem in the package itself, and the relevant code functionality was improved anyway in the PR #250

douglase, 2018-11-29:
no objections

closed 2018-12-04 by mperrin