spacetelescope/poppy #279: Option to return complex wavefront is not documented & hard-to-find

mperrin, 2018-12-10:
In https://github.com/spacetelescope/poppy/issues/234 (originally https://github.com/spacetelescope/mperrin/issues/234), @douglase added a feature suggested by @jlumbres: an option to return the complex wavefront instead of (in addition to?) returning the real-valued PSF. 

This is useful, but not at all documented anywhere. @sdwill had no idea this feature existed but it would have been useful to him.  Not immediately urgent but perhaps @sdwill, @douglase, @jlumbres, can some combination of you put together some brief docs on this, please?  @sdwill can take the lead on coordinating that into a PR eventually.

douglase, 2018-12-10:
There is some documentation of the feature in the description of [propogate_mono()](https://github.com/spacetelescope/poppy/blob/master/poppy/poppy_core.py#L1433), are you thinking it should be described in an example or elsewhere?

mperrin, 2018-12-10:
Some mention somewhere in the Sphinx docs perhaps? Not sure. I don't have a specific fix in mind.  

But demonstrably we should probably do _something_ since Scott asked for this feature and didn't realize you'd already implemented it...  (Then again, part of the problem is the gradual complexification of this API over time, which makes it harder to convey or even state all of what's in there.)

mperrin, 2018-12-10:
(More generally @douglase, it would be great if you could chat with @sdwill sometime. What you're both doing for WFIRST and LUVOIR modeling respectively has a lot of similarities, I think... )

mperrin, 2018-12-10:
One possible place to put this in the docs would be this page: https://poppy-optics.readthedocs.io/en/latest/options.html

But that's still got a great solution. In part because the docs could use a more substantial overhaul...