spacetelescope/poppy #289: Upstream astropy removal of astropy.tests.pytest_plugins

mperrin, 2019-02-11:
Apparently `astropy.tests.pytest_plugins` is moved someplace else in the next astropy. Not sure what change(s) we need to make in response to this. But right now our Travis builds are failing on the astropy development test case: 

See for instance this failing Travis build: 
https://travis-ci.org/spacetelescope/poppy/jobs/490316242

@shanosborne @robelgeda could one of you please take a look at this? @robelgeda have you seen anything similar with any of the other tools packages? Thanks.

mperrin, 2019-02-14:
@shanosborne @robelgeda did either of you get a chance to look at this yet? It would be nice to get it fixed, since the Travis builds are all being marked as failures right now because of this upstream change.

shanosborne, 2019-02-14:
@mperrin I've been out on vacation the past few days so I just saw this today. I'll get on it asap!

robelgeda, 2019-02-14:
Hi Marshall, I have a sprint this week but I see Shannon picked it up.

shanosborne, 2019-02-18:
I'm going to include some relevant links to this issue so that I can find them later if needed:

* Discussion of the timeline of the pytest_plugins issue by astropy people: https://github.com/astropy/astropy/issues/7782
* PR that closes the above issue and removes pytest_plugins: https://github.com/astropy/astropy/pull/7786
* Example fix for users pointed to in that PR: https://github.com/astropy/astroquery/commit/27f5c4d05f57b0e5871637ac4d936a64127007b4

PR #294 (2019-02-19): Fix astropy.tests.pytest_plugins error
Fixes #289 problem with astropy.tests.pytest_plugins. See links to reasoning behind this fix here: https://github.com/spacetelescope/poppy/issues/289#issuecomment-464885028
closed 2019-02-19 by mperrin