spacetelescope/poppy #297: Coveralls

robelgeda, 2019-02-25:
Coveralls is no longer reliable because github changed the way that 3rd party apps are integrated and coveralls appears to have done nothing to fix it. Its not clear if they will address this which means we should look for alternative services. One option that is currently popular is [Codecov](https://codecov.io). 

Related to #294 #296 
cc: @mperrin @shanosborne

mperrin, 2019-02-25:
I'm on board with switching to codecov. Is there recommended template setup for that we can lift from another ST package?

robelgeda, 2019-02-25:
I don't think ST has any templets for integration setup. I need to look into resources on how to.

mperrin, 2019-04-24:
@robelgeda have you been able to find any guidance for how to switch to codecov rather than coveralls? Do you think it's worth putting effort into this now, or leave it for some future time? Whichever you think best should be fine I think.

robelgeda, 2019-04-25:
I did not find anyone to help me with this. I think our best bet is to just try to do it ourselves. It would be a good learning experience as well I suppose. I am not sure if Coveralls is still not working so I will post on Slack to see what people say.

mperrin, 2020-02-04:
Codecov switch implemented in #349

closed 2020-02-04 by mperrin