spacetelescope/poppy #302: Digitization control

FredericCassaing, 2019-03-20:
Dear Marshall,
We would like to have a clear understanding about how poppy input parameters (which are actual physical values: wavelength, pixel angular size, fov, sampling factor, pupil diameter or face-to-face and gap) are converted to discrete arrays (total number of pixels, padding, pup/foc pixel size) in link of course with the sampling factor. We did not find direct information in the poppy documentation nor in the Optics Express 2007 paper. Our ultimate goal is to make an hexagonal-segmented aperture with a significant number of rings (4-10) with minimum distortion when digitally sampled, which can be obtained for specific values of the pupil_pitch/pupil_pixel ratio.
Before examining the code, may be there is a high-level document which describes that ? 
Many thanks in advance and best regards,
F. Cassaing

mperrin, 2019-03-20:
In general there is complete flexibility to make the pixel scale of the discrete arrays whatever you would like in both the pupil plane and image plane, independently of each other.  While the FFT imposes a fixed relationship between the sampling in one plane and the sampling in the other, the MFT is more general, and the transform parameters can be adjusted to go from any pupil plane scale to any desired image plane scale. (Still subject to the limitations of aliasing if you transform to a scale that is not Nyquist sampled, of course). This flexibility also allows the scales to be set independently of wavelength.   

Here is a simple example: 
```python
# define entrance pupil sampling at start of the optical system
osys = poppy.OpticalSystem(pupil_diameter=10*u.m, npix=1024, name='approximately_keck') 
osys.add_pupil(poppy.MultiHexagonAperture(rings=3, side=0.9*u.m))     
# define desired PSF sampling at the detector
osys.add_detector(pixelscale=0.01*u.arcsec/u.pixel, fov_pixels=50, oversample=1)  

# evaluate PSF and also return wavelengths so we can examine them
psf, wavefronts = osys.calc_psf(wavelength=1.5*u.micron, return_intermediates=True)
```

Because we set `return_intermediates=True`, we get back a `wavefronts` list that we can use to examine the sampling in each plane.  We can look at the sampling of the input pupil and find that the pixel scale is as expected = 10 meters / 1024 pixels: 
```
print(wavefronts[0].pixelscale)
0.009765625 m / pix
```
And in the output image plane, the pixel scale is as expected the specified 0.01 arcsec/pix:
```
print(wavefronts[1].pixelscale)
0.01 arcsec / pix
```
It should be straightforward to see how varying the input parameters changes those samplings.  If you have a specific desired discretization in the pupil, you can set that like this: 

```python
pupil_pixel_scale = 0.01*u.m/u.pixel
pupil_npix = 1200
osys = poppy.OpticalSystem(npix=pupil_npix, 
                           pupil_diameter=pupil_pixel_scale*pupil_npix*u.pixel)
```

For this kind of calculation using the MFT, padding for FFTs is not relevant and is not used. If you request an oversampled output, the MFT just directly transforms to the requested pixel scale from the entrance pupil (with no padding). Note, poppy uses different transforms in different contexts; the Fresnel code relies more heavily on FFTs so the considerations are somewhat different there than simple Fraunhofer imaging calculations, and zero-padding is used there. Some types of coronagraphic calculations can make use of different transforms as well, to optimize the sampling and calculation in various ways.   So depending on your application there may be other details we should discuss  - but for basic far-field imaging it's as simple as the above.

closed 2019-03-21 by FredericCassaing

FredericCassaing, 2019-03-21:
Dear Marshall,
thank you very much for this quick and complete answer. I closed the issue.
In the meantime we realized (but your comment is still very apreciated as it
explains how things work) that the bad positions we had are not linked to
bad integer truncations but rather seem linked to some kind of
mis-positionning from your code (cf issue #303).

By the way, since all relies on MFT, which can be called several times in
the same optical conditions (ie a constant MTF matrix but different pupil
fieds), is there a way to speed-up computations by pre-computing once for
all a static matrix of coefficients exp(i...) and then to rely on MKL to
just perform fast parallel matrix-vector products ?

Best regards,
-- 
Frédéric CASSAING - DOTA/HRA - HDR
Tél: +33 1 46 73 48 54
ONERA - The French Aerospace Lab - Centre de Châtillon
29 avenue de la Division Leclerc
BP 72 F-92322 CHATILLON CEDEX
http://www.onera.fr  |  Frederic.Cassaing@onera.fr
Avertissement/disclaimer http://www.onera.fr/onera-en/emails-terms

mperrin, 2019-03-21:
Replying to your question about speeding up the calculation - I have experimented a bit with code paths like that for acceleration. However the calculation of the coefficient arrays turns out not to be a significant part of the overall computation time. The matrix dot products are the large majority of the time. So in my tests I saw negligible speed ups from trying to precompute coefficients, and it does not seem to be worth the extra complexity in the code to do that. 

We’re certainly interested in speed improvements - see the SPIE paper last year by Ewan Douglas and myself on GPU acceleration of Fresnel calculations - but so far I have not found any good way to speed up the MFTs more. The MKL matrix code is already very, very good, and that is what needs almost the entire calculation time.