spacetelescope/poppy #303: MultiHexagonAperture - large gap related problem

LucasMarquis, 2019-03-21:
Dear Marshall,
We (FredericCassaing and I...) are faced with an alignment problem when calling the "poppy.MultiHexagonAperture" function (and, as a result, so does the "poppy.dms.HexSegmentedDeformableMirror" function). This is clearly visible when requiring a MultiHexagonAperture with a flattoflat/gap ratio lower than typical segmented pupils.  As you might notice on the attached picture, the ring number 3 is the only one concerned by a misalignment. It might match with the coding configuration of the poppy file "optics.py", line 1211, class MultiHexagonAperture, where the "_hex_center" function seems to give an individual code to those "C-like segments (in pairs)" (whatever the "# generalized code!" right below in this MultiHexagonAperture class).
We believe that the best approach would be to include the "ring=3" case in the general case of the following lines. Do you agree with this, or is there anything we have missed ?

[image]

mperrin, 2019-03-21:
Interesting! Thanks for this issue report. The explanation you propose seems likely to be correct, yes. Have you tested it by removing or commenting out the “rings=3” branch of the code? 

The history of that code is it was originally written in a less general way and only supported 1, 2, or 3 rings; then some time later I worked out the more generalized geometry. I suspect (but will check) that we can simplify that function to use the generalized code for all multi hex apertures of any size. 

If you want to make a pull request on github for this fix, please go ahead. Otherwise I will try to get to it sometime soon but perhaps not right away today.

mperrin, 2019-03-25:
PR submitted to fix this in #307

commit 2f45ade0 (2019-03-27): simplify MultiHexagonAperture; fixes #303; also bugfix for fixed_sampling_optic
closed 2019-03-27 by mperrin