spacetelescope/poppy #309: Broken display() method for HexSegmentedDeformableMirror() and MultiHexagonAperture()

ivalaginja, 2019-04-03:
I'm opening this issue after talking about this to @mperrin.

I wanted to display the OPD of a `HexSegmentedDeformableMirror()` object, but the result is a blank figure:

```py
iris = poppy.dms.HexSegmentedDeformableMirror(name='Iris DM',
                                              rings=2,
                                              flattoflat=1.44 * u.mm,
                                              gap=10 * u.micron)
iris.display(what="both")   # this does not work
plt.show()
```
[image]

This is partly true for a `MultiHexagonAperture()` object - they inherit the same `display()` method:
```
ap = poppy.MultiHexagonAperture(rings=2, flattoflat=flat, gap=flat/10)
ap.display()
plt.show()
```
[image]
The transmission is fine, but the OPD should actually be blank here, since it is zero everywhere, not equivalent to the transmission.

When accessing the transmission and/or OPD separately with a wave, this works fine for the `MultiHexagonAperture()` object:
```py
flat = 1*u.m
w = poppy.Wavefront(diam=6*flat, npix=512)
ap = poppy.MultiHexagonAperture(rings=2, flattoflat=flat, gap=flat/10)

plt.figure()
plt.subplot(1, 2, 1)
trans = ap.get_transmission(w)
plt.imshow(trans)  # this does work
plt.title("Transmission")

plt.subplot(1, 2, 2)
opd = ap.get_opd(w)
plt.imshow(opd)
plt.title("OPD")
plt.show()
```
[image]

And also for the `HexSegmentedDeformableMirror()`, except that the aperture within the array scales inversely with its physical size `flat`, which should not be the case and is also not happening in the above `MultiHexagonAperture() case`:
```py
iris = poppy.dms.HexSegmentedDeformableMirror(name='Iris DM',
                                              rings=2,
                                              flattoflat=1.44 * u.mm,
                                              gap=10 * u.micron)
flat = 0.001*u.m    # the bigger this number, the smaller the display??
w = poppy.Wavefront(diam=6*flat, npix=512)

plt.figure()
plt.subplot(1, 2, 1)
trans = iris.get_transmission(w)
plt.imshow(trans)   # this works, but weird sizing
plt.title("Transmission")

plt.subplot(1, 2, 2)
opd = iris.get_opd(w)  # this works too, but weird sizing
plt.imshow(opd)
plt.title("OPD")
plt.show()
```
[image]

Example of weird aperture sizing in array:
```py
iris = poppy.dms.HexSegmentedDeformableMirror(name='Iris DM',
                                              rings=2,
                                              flattoflat=1.44 * u.mm,
                                              gap=10 * u.micron)
plt.figure()
plt.subplot(1, 3, 1)
flat = 0.001*u.m    # the bigger this number, the smaller the display??
w = poppy.Wavefront(diam=6*flat, npix=512)
trans = iris.get_transmission(w)
plt.imshow(trans)   # this works, but weird sizing
plt.title("flat = 0.001*u.m")

plt.subplot(1, 3, 2)
flat = 0.01*u.m    # the bigger this number, the smaller the display??
w = poppy.Wavefront(diam=6*flat, npix=512)
trans = iris.get_transmission(w)
plt.imshow(trans)   # this works, but weird sizing
plt.title("flat = 0.01*u.m")

plt.subplot(1, 3, 3)
flat = 0.1*u.m    # the bigger this number, the smaller the display??
w = poppy.Wavefront(diam=6*flat, npix=512)
trans = iris.get_transmission(w)
plt.imshow(trans)   # this works, but weird sizing
plt.title("flat = 0.1*u.m")
plt.show()
```
[image]

The most recent merged PRs touching the `MultiHexagoonAperture()` are PR #307 and PR #306 , although only the latter seems to have changed something on the `display()` method. I wasn't able to track this down fully.

mperrin, 2019-04-04:
@robelgeda and maybe @joseph-long, would either of you maybe have time to look at this issue?

mperrin, 2019-04-04:
Not sure if this could be my bug from #307, @joseph-long's bug from #306, or some totally unrelated latent bug that's been around for a while but didn't crop up until now.

mperrin, 2019-04-23:
OK, I've tracked down the root cause of this, though am still not sure of the best fix. This is a side effect related to @joseph-long's PR #306. Specifically the changes around https://github.com/spacetelescope/poppy/blob/master/poppy/poppy_core.py#L2457
And the problem is much broader than just the classes named here; in fact it affects _all_ analytic optical element classes. 

In that PR, as part of the change to allow wavelength-independent phase maps, @joseph-long updated the display method to call `get_transmission` and `get_opd` on a Wavefront instance, rather than just accessing the `.amplitude` and `.opd` attributes. This is in conflict with how the display code in `optics.py` for AnalyticOpticalElement works, which is by sampling the optic function into arrays stored in `.amplitude` and `.opd` temporarily. 

I may have to substantially overhaul/refactor the display code to make this work correctly in all instances. Worth doing and I can probably clean things up while doing so. But it continues to surprise and dismay me how much of the complexity in this codebase is related to display and graphics... Sometimes it feels that's even harder than getting the physics right!

commit 73628d00 (2019-05-08): Refactor optic display to fix #309. Significant change to how this works for analytic optics, perhaps overall a simplification.
closed 2019-05-08 by mperrin