spacetelescope/poppy #310: Problem with building Fresnel Optical system with ZernikeWFE in place

jlumbres, 2019-04-11:
I'm simulating a Fresnel Optical system with some aberration in place (such as a tilted source) using the ZernikeWFE function. However, I'm unable to build the wavefronts when I use calc_psf. It seems there's a problem in calc_psf to take in the FresnelWavefront? 

The Fresnel propagator with the ZernikeWFE has worked for me in the past with version 0.6, but I've since updated to the recent developer version ('0.8.1dev1324.dev1324'). I'm opening this error since I've talked with @douglase on this issue.

Here's sample code I generated:

```
# Make a basic optical system
EP_diam = 10e-3 * u.m
oversamp = 0.125
samp = 256
WAVELEN = 632.8e-9*u.m
fl_OAP = 0.2*u.m

# Initialize FresnelOpticalSystem
testbed = poppy.FresnelOpticalSystem(pupil_diameter = EP_diam, npix=samp, 
                                     beam_ratio=oversamp)

# First surface is the entrance aperture.
testbed.add_optic(poppy.CircularAperture(radius=EP_diam/2))

# Add in some wavefront error from source
aberr_coeff_sequence = [0, 0, 0, 0.9]
# Declare LGS source for testbed
sys_aberr = poppy.ZernikeWFE(radius=EP_diam.value/2, coefficients=LGS_coeff_sequence)
testbed.add_optic(sys_aberr)

# OAP1
testbed.add_optic(poppy.QuadraticLens(fl_OAP, name='OAP-1'))
testbed.add_optic(poppy.CircularAperture(radius=0.025*u.m,name="OAP1 aperture"))

# Focal plane
testbed.add_optic(poppy.ScalarTransmission(planetype=PlaneType.intermediate, 
                                           name="focal plane"), distance=fl_OAP)

fp_psf, fp_wfs = testbed.calc_psf(wavelength=WAVELEN.value, 
                                        display_intermediates=True, 
                                        return_intermediates=True)
```
But then I get a very long-winded error that does not produce wavefronts. Here's a copy of the error at the end and the aperture I produced before the wavefront error was implemented:
```
ValueError: Argument 'wavelength' to function '__init__' must be a number (not 'Wavefront:
        wavelength = 0.6328 micron
        shape = (2048, 2048)
        sampling = 3.90625e-05 m / pix'), and convertable to units='m'.
```
[image]

Any comments on how to approach this?

mperrin, 2019-04-11:
Thanks! FYI we likely won’t be able to look at this or #311 until next week, given an ongoing JWST WFSC rehearsal this week. But definitely thanks for reporting this.

mperrin, 2019-04-24:
With the PR #311 merged in now, I've verified that the above code runs and outputs a PSF, without getting any errors.   @jlumbres, @douglase would you concur with marking this issue closed now? 

(Minor caveat, to get a reasonable PSF I had to change the value scaling for `aberr_coeff_sequence` above, which right now specifies 0.9 _meters_ of focus, presumably accidentally :-)

jlumbres, 2019-04-26:
I downloaded the new version, it works! Thanks!

closed 2019-04-26 by jlumbres