spacetelescope/poppy #318: Indexing error in get_opd() of HexSegmentedDeformableMirror() class when center=False

ivalaginja, 2019-04-30:
Discovered while testing PR #314 I found a bug in the plotting of a `HexSegmentedDeformableMirror()` instance. When I define it with `center=False`, meaning I omit the center segment, the loop in `get_opd()` still tries to address the full number of segments:

```py
import poppy
import astropy.units as u
import matplotlib.pyplot as plt
%matplotlib inline

iris = poppy.dms.HexSegmentedDeformableMirror(name='Iris DM',
                                              rings=3,
                                              flattoflat=15.*u.m,
                                              gap=2*u.cm,
                                              center=False)
plt.figure(figsize=(16, 8))
iris.display(what='both')
```

Error message:
```
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-3-56c9b836c811> in <module>()
     11                                               center=False)
     12 plt.figure(figsize=(16, 8))
---> 13 iris.display(what='both')
[...]
~/anaconda/envs/pastis/lib/python3.6/site-packages/poppy/dms.py in get_opd(self, wave)
    594             wseg = self._seg_indices[i]
    595             self.opd[wseg] = (self._surface[i, 0] +
--> 596                               self._surface[i, 1] * self._seg_x[wseg] +
    597                               self._surface[i, 2] * self._seg_y[wseg])
    598         return self.opd

IndexError: index 36 is out of bounds for axis 0 with size 36
```

It works fine with `center=True`:
```python
iris = poppy.dms.HexSegmentedDeformableMirror(name='Iris DM',
                                              rings=3,
                                              flattoflat=15.*u.m,
                                              gap=2*u.cm,
                                              center=True)
plt.figure(figsize=(16, 8))
iris.display(what='both')
```

ivalaginja, 2019-05-03:
This might be easily fixed by adjusting `self.segmentlist` in its base class `MultiHexagonAperture()` depending on whether we have `center=True` or `center=False`. Would need to look into it in more detail though.

mperrin, 2019-05-03:
Great, if you want to test it out and hopefully submit a PR with a fix that would be great. I expect you are correct and something like that is the necessary fix.

ivalaginja, 2019-05-03:
Turns out that `self.segmentlist` is behaving as you would expect, it simply omits the `0` if you tell it not to include the center segment.
The problem arises because `self._surface` is an array while `self._seg_indices` is a dictionary. The dictionary keeps track of the segment numbers all right because it is not ordered, but `self.surface` is ordered and hence accesses its entries wrong when asked for a specific segment.

More about this over on PR #319.

PR #320 (2019-05-09): allocate sufficient-sized _surface array even if missing center segment. Fixes #318
An alternate way to fix #318. See discussion in comments on #319. 

Attention @ivalaginja
commit e34e01ee (2019-05-09): allocate sufficient-sized _surface array even if missing center segment. Fixes #318
closed 2019-05-09 by mperrin