spacetelescope/poppy #328: Remove use of astropy-helpers submodule

mperrin, 2019-06-17:
See astropy/astropy-APEs#52 ("A roadmap for package infrastructure without astropy-helpers") which describes how the functionality in astropy-helpers can be achieved via more standardized tools in the Python ecosystem. 

This is clearly good and we should do it. 

See for instance https://github.com/spacetelescope/mosviz/pull/179 and https://github.com/spacetelescope/cubeviz/pull/492 for examples of PRs that remove astropy-helpers. We should do similarly here.

joseph-long, 2019-07-15:
First of all, 🙌 

Second of all, I recently got an email from a would-be user that had copied the `poppy` dir into their Python path without going through setuptools, which skipped `version.py` generation, which made them think POPPY was broken. Just another data point against clever things in `setup.py` 😉

mperrin, 2019-07-15:
I think I just wrote back to the same user - I had an email describing that problem with version.py waiting in my inbox when I got back from vacation. Were you able to get them up and running already?

joseph-long, 2019-07-15:
Yes, I directed him to use the issue tracker in the future :)

joseph-long, 2019-07-15:
(And he reports that everything worked after following the installation instructions to the letter)

pllim, 2020-01-24:
You might want to do this sooner than later (or update your existing `astropy-helpers` version). @bhilbert4 reported this error trying to install `poppy` as part of `mirage` CI:

```
Searching for poppy>=0.9.0
Reading https://pypi.org/simple/poppy/
Downloading https://files.pythonhosted.org/packages/5e/dc/6799b3048282b37bb86fcc6ed1aa22d7c80d3264e92441153d02001ccb5e/poppy-0.9.0.tar.gz#sha256=43ccfb013690311022fe47626c503f2f7dbdb5aabe9f5df15ee828510088aa92
Best match: poppy 0.9.0
Processing poppy-0.9.0.tar.gz
Writing /tmp/easy_install-egmpb_3y/poppy-0.9.0/setup.cfg
Running poppy-0.9.0/setup.py -q bdist_egg --dist-dir /tmp/easy_install-egmpb_3y/poppy-0.9.0/egg-dist-tmp-389jg3ze
Sorry: TabError: inconsistent use of tabs and spaces in indentation (get_psf.py, line 19)
/tmp/easy_install-egmpb_3y/poppy-0.9.0/.eggs/astropy_helpers-2.0.11-py3.7.egg/astropy_helpers/setup_helpers.py:102: AstropyDeprecationWarning: Direct use of the adjust_compiler function in setup.py is deprecated and can be removed from your setup.py.  This functionality is now incorporated directly into the build_ext command.
  'command.', AstropyDeprecationWarning)
error: SandboxViolation: open('/home/docs/.astropy/config/astropy.4.0.cfg', 'wt') {'encoding': 'latin-1'}
The package setup script has attempted to modify files on your system
that are not within the EasyInstall build area, and has been aborted.
This package cannot be safely installed by EasyInstall, and may not
support alternate installation locations even if you run its setup
script by hand.  Please inform the package's author and the EasyInstall
maintainers to find out if a fix or workaround is available.
```

If you have already fixed this on dev, then you might need to do a new release. FYI.

mperrin, 2020-01-24:
@shanosborne - do you have any availability to work on the de-astropy-helpering poppy and webbpsf? I believe this is now becoming more urgent with the latest astropy release. See above comment by @pllim.

pllim, 2020-01-24:
@mperrin , just to be clear, `astropy` itself has not removed `astropy-helpers` yet (slated for 4.1). For your situation, simply updating `astropy-helpers` version to its latest tag might fix the issue. Although, if you are putting in the work already, might as well go all the way and apply APE 17? Tom R wrote a draft of a guide [here](https://demo.codimd.org/6TnoSu6PS1m-H8socv7JDg) in case you find it helpful.

bhilbert4, 2020-01-24:
To be clear, I'm only seeing the error above during the Readthedocs build for Mirage. When installing Mirage, everything seems to build ok.

pllim, 2020-01-24:
@bhilbert4 , can you please link the log to the failed doc build?

bhilbert4, 2020-01-24:
https://readthedocs.org/projects/mirage-data-simulator/builds/10317921/

shanosborne, 2020-01-24:
@mperrin yes I can get back to this issue!

bhilbert4, 2020-01-24:
Mirage RTD builds are now passing. I removed a bunch of dependencies, including webbpsf/poppy  from the build, and everything looks good.

mperrin, 2020-02-04:
Implemented in #349

PR #349 (2020-02-04): Remove astropy-helpers and implement codecov
This PR attempts to remove the dependency on the `astropy-helpers` sub-package. The big changes are:
- All `setup.py` options have been moved to `setup.cfg` (with some over-arching requirements set in `pyproject.toml`
- All `travis` options have been moved to `tox.ini` which now controls testing, doc building, and coverage.
- Coverage is now run through `codecov` rather than `coveralls`
- `_astropy_init.py` has been removed and relevant information has been moved to `__init__.py`
- Updated docs structure so they would build correctly

Closes #328 an [... cut]
closed 2020-02-04 by mperrin