spacetelescope/poppy #332: radial_profile function does not use energy relative to detector

ariedel, 2019-07-30:
The utils.measure_radius_at_ee function is meant to provide a function that yields the radius of a given amount of encircled energy. 
However, as currently calculated, that number (from utils.radial_profile) is based on summing up the flux... which is normalized to the pupil plane, not the detector plane.

If the flux were distributed uniformly, a circle of radius r should have 75.5% of the area (and flux) of the square (of side 2r) that circumscribes it.
What this means, practically, is that for a decent sized postage stamp PSF (such as used by Pandeia), I have seen the largest EE measured on the image by utils.radial_profile() be as low as 75.8%, which implies there's an enormous amount of flux in the corners of the PSF, which is mainfestly not true.

Proposed solution: normalize the cumulative sum of the pixels, to the sum of the PSF. I believe it could be done on lines 595 and 603 of utils.py.

mperrin, 2019-08-06:
Hi @ariedel I thought I had commented on this issue before but apparently it didn't get posted somehow...

The normalization in the `radial profile` function is intentional. Normalizing to the pupil intensity allows for the fact that some significant amount of flux can end up outside any small postage stamp. In the case of the size of postage stamps used in Pandeia (typically a couple arc seconds, yes?) it can indeed be the case that 5-10% or more of the incident flux can be diffracted to outside that postage stamp area. (The surface brightness flux in any such pixel is quite low, but there are a _lot_ of pixels out there at large separation... There can be an unintuitively large fraction of light scattered to wide separations.)

A separate factor can be cases in which much of the light is blocked by some mask, such that it never makes it to the PSF at all (e.g. AMI mask, coronagraph Lyot stop, spectrograph slit). It's a subtle question whether the EE should be normalized relative to that or not. If you care about fractional encircled energy just relative to the whole PSF then it does make sense to normalize to the sum of the PSF; on the other hand if you care about absolute photometry then it doesn't. Since this depends on the usage scenario we left it up to the user. 

That said, there's an existing option to do this. Just set `normalize="total"` in a call to `radial_profile` and you'll get the behavior that you want.   But I see that keyword isn't supported by `utils.measure_radius_at_ee`. Would it suffice for your purposes if we added a `**kwargs` pass-through so that you can use that option with `measure_radius_with_ee`?

ariedel, 2019-08-06:
Yes, that would suffice.

closed 2019-08-08 by mperrin