spacetelescope/poppy #355: pip install poppy fails because it seems to require astropy-helpers 2.0.11

robelgeda, 2020-03-19:
@larrybradley  wrote:

To close the loop on not being able to install webbpsf via pip.  I investigated further, and the real issue is with the required poppy package.  In a clean environment, pip install poppy fails because it seems to require astropy-helpers 2.0.11, which is an ancient version:
```
Exception: Error retrieving astropy-helpers from PyPI:
    DistutilsError("Command '['/Users/lbradley/.venvs/psf-nb/bin/python3', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/var/folders/ly/xztygm217pd97f4m_4srfbw0000132/T/tmpdnbh5jr9', '--quiet', '--index-url', 'https://pypi.python.org/simple', 'astropy-helpers==2.0.11']' returned non-zero exit status 1.")
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

If I manually do pip install 'astropy-helpers==2.0.11' , I can then pip install poppy and webbpsf.  So, there seems to be some issue with the setup in poppy with regards to astropy-helpers  (which is no longer recommended for packaging).

mperrin, 2020-03-19:
Thanks @robelgeda and @larrybradley. The development version of poppy no longer requires astropy helpers at all, thanks to @shanosborne. 

I think we have not been planning a new release immediately (due to the broader desire to minimize new tool software releases in the run up time before the JWST deadline.) But we could revisit that, and in particular could consider a small delta release for poppy and webbpsf if you think that’s warranted.

robelgeda, 2020-03-19:
This sounds good to me

mperrin, 2020-03-20:
In that case @robelgeda can you please coordinate with @shanosborne to organize for an 0.9.1 release of poppy?  This can just be a straightforward release of all the changes made since 0.9.0 in poppy. It doesn't sound like a webbpsf release is needed at this time; do you agree?

mperrin, 2020-08-04:
This was resolved with the last release. Closing issue.

closed 2020-08-04 by mperrin