spacetelescope/poppy #362: Migrate from pysynphot to synphot

mperrin, 2020-06-16:
Synphot is a refactor of pysynphot and we should start using the newer package. This is recent work done by @ariedel in support of the ETC (see discussion at [this innerspace page](https://innerspace.stsci.edu/pages/viewpage.action?spaceKey=JWST&title=2020-06-15+ETC+release+schedule+update%2C+new+template+spectra+and+filter+bandpasses%2C+synphot+refactor) and [this presentation by Adric](https://innerspace.stsci.edu/download/attachments/220693357/Synphot_Presentation.pptx?version=1&modificationDate=1592254666182&api=v2) and [this Pandeia PR](https://github.com/spacetelescope/pandeia/pull/4949).

We should update poppy (and webbpsf) to also use synphot instead of pysynphot.

pllim, 2020-08-03:
This might also enable #373

mperrin, 2021-02-04:
Not sure if there is a particular priority for this or not - is pysynphot being deprecated at some point or not, is there any concrete benefit to switching, etc?  I have to admit the relationship between pysynphot, synphot, and stsynphot is not entirely clear to me.

ariedel, 2021-02-04:
The ETC team has heard that DATB would prefer to stop maintaining pysynphot, now that there's a new astropy-affiliated package they'd prefer to expend their effort on. We were told that we were the major users, within the institute, of pysynphot.

synphot is the generic parts of pysynphot - loading, normalizing spectra, and the like.
stsynphot is all of the institute-specific parts (anything that touches CDBS data or the graph table, like the ability to look up HST filters and dispersers by obsmode string; convenience functions for some additional flux types like PHOTLAM; and to load interpolated spectra out of model grids in CDBS).

pllim, 2021-02-04:
Yes, PySynphot is in legacy mode and will be archived once HST ETC stop using it. Its roadmap is documented at https://github.com/spacetelescope/pysynphot/blob/master/ROADMAP.md .

You might find the following conversion guides useful:

* https://synphot.readthedocs.io/en/latest/synphot/from_pysyn_iraf.html
* https://stsynphot.readthedocs.io/en/latest/stsynphot/from_pysyn_iraf.html

I am looking into how you are using PySynphot in your code and I plan to submit an exploratory PR soon. FYI.

PR #388 (2021-04-01): MNT: Port from pysynphot to stsynphot
Fix #362 

cc @mperrin 

**TODO**

- [ ] Have stakeholders review and test this.
- [x] Fix code and tests, if necessary.
closed 2021-04-01 by mperrin