spacetelescope/poppy #363: Potential bug in poppy.measure_fwhm

kjbrooks, 2020-06-17:
In attempting to use poppy.utils.measure_fwhm, I ran into a TypeError: No loop matching the specified signature and casting was found for ufunc true_divide. 

It seems that this line: 

https://github.com/spacetelescope/poppy/blob/2010aaac2e738ac347186e28b2258e489a22deec/poppy/utils.py#L821

is causing an error since this operation is not actually valid for the reasons laid out in this issue: https://github.com/numpy/numpy/issues/10565/

It seems like a relatively simple solution of change that line to: `image = image / image.max()`

mperrin, 2020-06-18:
Oh interesting. This is for an image with integer dtype? I think it sounds like  the existing code works for float but not for int? Good catch!

kjbrooks, 2020-06-18:
Yes, this was an image that came out of MAGIC and is dtype uint16

PR #368 (2020-06-25): Update utils.py so that measure_FWHM works on integer as well as float dtype
If the input fits image is integer dtype, the current code breaks. Making this change will make this work for both float and int arrays.

Closes #363
commit 39260eb2 (2020-06-25): Update utils.py

If the input fits image is integer dtype, the current code breaks. Making this change will make this work for both float and int arrays.

Closes #363
commit c05e0f2e (2020-06-25): Merge pull request #368 from spacetelescope/kjbrooks-patch-1

Update utils.py so that measure_FWHM works on integer as well as float dtype. Closes #363
closed 2020-06-25 by mperrin