spacetelescope/poppy #375: How to propagate 2D image wavefront

joncox123, 2020-08-10:
Can poppy propagate a 2D image wavefront, such as an arbitrary bitmap image, or does it only compute the PSF? If the later, then I presumably need to implement the code to convolve the image with PSF? I couldn't find a routine or example for loading a bitmap image as a wavefront.

By the way, for us boneheads, what's the relationship between focal length and waves of defocus for a thin lens?

douglase, 2020-08-10:
If you represent your wavefront as an OPD and an amplitude mask, you can load it using the FITS optical element function, multiply it by a flat wavefront and then propagate it to the image plane, no post-facto convolution required: https://poppy-optics.readthedocs.io/en/stable/extending.html#defining-a-custom-optic-from-a-fits-file

telescope-optics.net has a nice discussion of defocus error:
https://telescope-optics.net/defocus1.htm#error

joncox123, 2020-08-10:
@douglase thanks for the instructions. Regarding the derefocus question, although I am also interested in this aberration, the first thing I need to do is simply to simulate a thin lens with a certain focal length. However, the ThinLens object in Poppy only takes a defocus term, not a focal length. 

Is my understanding correct that the user must specify the focal length somehow in terms of waves of defocus? It’s not possible to just enter a focal length of a thin lens in units such as meters (e.g f = 100 mm lens)?

Admittedly, I’m not really an imaging expert, so the methodology is different to me. I guess what Poppy really does is stimulate only the aberration part, not the normal optical propagation. So you can’t enter f = 100 mm for a thin lens, only the delta in f = 100 mm * delta. Right? If so, I guess I have to think about whether this C will enable the full optical imaging stimulation I need.

Maybe the docs can just explain it in terms of what your typical physics PhD is used to?

douglase, 2020-08-10:
Since the Fraunhofer approximation is far field, in the Fraunhofer imaging regime there are no focal lengths, just pupil and image planes, so the thin lens class in POPPY only serves as a way to add the defocus aberration. Depending on the regime you are in, you might not need to the Fresnel module, but it does account for each elements' focal length. The Hubble example of Fresnel propagation under the "FresnelOpticalSystem" headin in this notebook: https://github.com/spacetelescope/poppy/blob/develop/notebooks/Fresnel_Propagation_Demo.ipynb should be all you need to get started but if you'd like more examples of real optical systems, this repository has simulations of a few complex optical benches at UArizona with POPPY: https://github.com/jlumbres/UAWFSTB/blob/master/LGS_offAxisModel.ipynb

citation: Lumbres, J., Males, J., Douglas, E., Close, L., Guyon, O., Cahoy, K., et al. (2018). Modeling coronagraphic extreme wavefront control systems for high contrast imaging in ground and space telescope missions. In Adaptive Optics Systems VI (Vol. 10703, p. 107034Z). International Society for Optics and Photonics. https://doi.org/10.1117/12.2313780

the docs have an undeniable astronomy bent since most of us are astronomers, feel free to open pull requests if you see places to de-jargon things.

closed 2021-01-05 by mperrin