spacetelescope/poppy #381: wave is included in get_opd() but not used

douglase, 2020-10-16:
https://github.com/spacetelescope/poppy/blob/71dddf818ca600c771235e54473473426b0b90a2/poppy/poppy_core.py#L2302

mperrin, 2020-10-16:
This is intentional as part of the API, for multiple reasons. It allows wavelength-dependent OPDs in subclasses for refractive optics. And besides we use the wave’s sampling to rasterize the optic’s OPD values for any of the analytic optic subclasses.

douglase, 2020-10-16:
Apologies for terse description. The particular line was causing confusion,
I was trying to help another user access their already generated .opd in
meters, would it be appropriate for this realization of .get_opd() to have
a comment in the docstring that the wavelength is not needed and .opd will
be returned?

mperrin, 2020-10-16:
Sure, how about something like this in the doc string: 

"In this base class instance, the wavefront parameter 'wave' is not used, and the .opd attribute is returned directly. Subclasses may change this behavior, for instance to evaluate optical aberrations on the sampling defined for that wavefront, or to compute wavelength-dependent aberrations of refractive optics."

And, for the case you're describing, it sounds like they can also just access the .opd attribute directly, and avoid calling this function at all. That's an option too...

douglase, 2020-10-16:
yep, that's what we ended up going with, #382 adds a slightly tweaked version of your docstring above.

closed 2020-10-16 by douglase