spacetelescope/poppy #405: extract an image of a pupil plan

Teusia, 2021-03-02:
Dear all,

I am trying to setup a phasing wavefront sensor simulation using poppy. It happen that for some wavefront sensors I need to extract an image of the pupil plan. I know poppy can calculate this image because when doing poppy.OpticalSystem.calc_psf(display _intermediates = True) then .display_psf there is a beautiful pupil image showing me the intensities I expect. However, I cannot figure out how to extract this image, how can I do this please?

Here is some example code to show you what I did:
```python
import poppy
import matplotlib.pyplot as plt
import numpy as np
import astropy.units as u

plt.figure(1)
plt.clf()
hexgmt = poppy.dms.HexSegmentedDeformableMirror(rings = 1, flattoflat=9.698*u.meter,gap = 0.2*u.meter)
hexgmt.set_actuator(1,0.249*u.micron, 0 , 0)

pm = poppy.CircularPhaseMask(name = 'zeus',radius = 0.05*u.arcsec, retardance = 0.25)

plt.figure(2)
plt.clf()
osys = poppy.OpticalSystem()

osys.add_pupil(hexgmt)

osys.add_image(pm)
osys.add_pupil()
osys.add_detector(pixelscale = 0.01, fov_arcsec = 5.0)
psf = osys.calc_psf(wavelength = 1e-6, display_intermediates=True)
poppy.display_psf(psf,colorbar = False)
```
This code produces the image that follow and I would like to extract the image I surrounded in yellow, please?

[image]

mperrin, 2021-03-02:
Hi, this is possible using the `return_intermediates` parameter to calc_psf. Like: 

```python
psf, waves = osys.calc_psf(wavelength = 1e-6, display_intermediates=True, return_intermediates=True)

plt.figure()
waves[2].display()
```
The `waves` variable will be a list of complex wavefronts, one for each plane in the optical system. You can then access, for instance, `waves[2].intensity` for the intensity at that pupil plane.

mperrin, 2021-03-02:
The other way to do this is to set a detector instance in the pupil plane. This allows setting the pixel scale to a particular value, making polychromatic pupil images, and so on.  I think the documentation may not show an example of a detector in a pupil plane but it can be done.  

Note also, for simulations of Zernike sensors it can be important to set the numerical sampling of the first image plane to a high spatial resolution, to better resolve the PSF and Zernike mask. I expect either @raphaelpclt or @ivalaginja could help with examples for how to do this. I can help also, but am I correct you are at Marseille so may know them?

ivalaginja, 2021-03-02:
Hey! Yes we know each other, it was in fact me who directed her to the poppy docs and the repo ^^
We can certainly figure out how to put a detector in the pupil plane, but extracting the correct intermediate plane like in the example @mperrin posted should be a good start. It depends very much on how you intend to use that information further that might motivate the one or the other, at least that's how it's usually been for me.

Teusia, 2021-03-02:
Hello,

Great thank you!!!
:) I am no longer in Marseille, but yes I know Iva, not sure about Raphael though.
Based on your comment @ivalaginja  I shall start with the example of @mperrin.

On the longer term being able to control the pupil plan sampling, and poly-chromatic images, would definitively be a plus. My intention is to use the produced images to develop an image processing algorithm. The later algorithm will be intended to also work with images produced by a test bench.

Would this require further development?

mperrin, 2021-03-02:
@raphaelpclt is working on Zernike sensors nearby at Observatoire de la Cote d'Azur. We are using poppy to simulate data and develop analysis algorithms for use with our HiCAT test bench Zernike sensor; it sounds like what you're interested is quite similar, so yes it can be done.  

What's needed is not new development in poppy, but setting the appropriate options in the calculation setup, for instance to use the MTF.

raphaelpclt, 2021-03-03:
Hello :) 
If I had to take bets, someone called Anne-Laure who was in Marseille and who works on segmented apertures, I guess we know each other ^^ In fact we did some experiments on MITHiC together.

To come back on @mperrin 's comment, indeed I do use poppy detectors. I did not spend time on understanding why exactly, but it seems I need to do it in 2 seperated steps. First define the detector object:
```
your_detector = poppy.Detector(name='detector', pixelscale=cam_info['pixelscale'], fov_pixels=cam_info['height'])

```
and then add it to the optical system:

`osys.add_pupil(your_detector)`

I found out that using osys.add_detector() sees its scaling and field of view overridden by some other propagation parameters. Again, I did not go to far in understanding this, maybe it's just a HiCAT-related feature.

Teusia, 2021-03-04:
:) 
Thank you Raph, that is exactly what I needed!
For the MTF I' ll try to find again the documentation, and if anything comes up I'll ask.
Many thanks!!

closed 2021-03-04 by Teusia