spacetelescope/poppy #410: poppy for windows users and git

Teusia, 2021-03-23:
Hello,

So I had to verify that my windows colleagues can use poppy, and it turns out that the setup.py of poppy requires git. The windows machine I tested on had git Desktop installed, which seam not recognised by python. Because I am using Anaconda to manage python I could solve this by an anaconda install of git  (e.g.: `conda install git` there is also a python-git in pypi). 
Maybe this would be worth mentioning in the installation instruction?
However is there really a fundamental need for git in the poppy installation?
I think this is somewhat important to mention because the most common stack overflow first answer for the error raised when poppy install fails (which is `OSError: 'git' was not found`), is to add a PYTHONPATH to windows and add the path to a installed version of git. This breaks Anaconda!!!

Cheers,
Anne-laure Cheffot

mperrin, 2021-03-23:
Hi @Teusia - interesting, thanks for this report. I have never tried to install or use poppy on Windows without git.  Windows is only semi-supported and not well tested (since few astronomers seem to use Windows these days!) but I agree it should be possible to install poppy on Windows.  Anaconda is our (STScI's) supported and recommended way to install Python packages. Is it not possible for your colleagues to also use Anaconda? 

All that said, I agree there should not be a fundamental need for git in the poppy installation, not if installing from PYPI via pip. On the other hand if installing from Github then yes git is required, as a dependency coming in from the use of setuptools_scm for version information, which happens when installing the source from github.   

Can you please provide more information on the specific error messages being output, and whether your colleagues are trying to install from pypi via pip or not. My colleague @shanosborne may also be able to help with some of these installation issues.

Teusia, 2021-03-23:
Hello,

I am still in the process of testing so please do use me to ask for everything you require to help with this.
### install with anaconda
Anaconda is also our python package manager but on anaconda poppy seem to only be [supported for osx](https://anaconda.org/stsci/poppy). Trying to install nonetheless yield:
```
$ conda install poppy
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - poppy

Current channels:

  - http://ssb.stsci.edu/astroconda/win-64
  - http://ssb.stsci.edu/astroconda/noarch
  - https://repo.anaconda.com/pkgs/main/win-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/win-64
  - https://repo.anaconda.com/pkgs/r/noarch
  - https://repo.anaconda.com/pkgs/msys2/win-64
  - https://repo.anaconda.com/pkgs/msys2/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```
BTW this is the case for the entire stsci astroconda install it seems windows is just not supported [see the astroconda system requirement](https://astroconda.readthedocs.io/en/latest/)

### Install with pip from pypi
I tried the pip installation of poppy that works wonders, it does all the dependencies and everything.

### install with setup.py
The one that really does not work upfront is when cloned from the git repository and try to install from the anaconda powershell:
```
(testenv) PS ~\simulations\python\poppy> python setup.py install
Traceback (most recent call last):
  File "~\simulations\python\poppy\setup.py", line 53, in <module>
    setup(use_scm_version={'write_to': os.path.join('poppy', 'version.py')})
  File "~\anaconda3\envs\testenv\lib\site-packages\setuptools\__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "~\anaconda3\envs\testenv\lib\distutils\core.py", line 108, in setup
    _setup_distribution = dist = klass(attrs)
  File "~\anaconda3\envs\testenv\lib\site-packages\setuptools\dist.py", line 432, in __init__
    _Distribution.__init__(self, {
  File "~\anaconda3\envs\testenv\lib\distutils\dist.py", line 292, in __init__
    self.finalize_options()
  File "~\anaconda3\envs\testenv\lib\site-packages\setuptools\dist.py", line 708, in finalize_options
    ep(self)
  File "~\anaconda3\envs\testenv\lib\site-packages\setuptools\dist.py", line 715, in _finalize_setup_keywords
    ep.load()(self, ep.name, value)
  File "~\simulations\python\poppy\.eggs\setuptools_scm-6.0.1-py3.9.egg\setuptools_scm\integration.py", line 24, in version_keyword
    dist.metadata.version = _get_version(config)
  File "~\simulations\python\poppy\.eggs\setuptools_scm-6.0.1-py3.9.egg\setuptools_scm\__init__.py", line 173, in _get_version
    parsed_version = _do_parse(config)
  File "~\simulations\python\poppy\.eggs\setuptools_scm-6.0.1-py3.9.egg\setuptools_scm\__init__.py", line 127, in _do_parse
    version = _version_from_entrypoints(config) or _version_from_entrypoints(
  File "~\simulations\python\poppy\.eggs\setuptools_scm-6.0.1-py3.9.egg\setuptools_scm\__init__.py", line 65, in _version_from_entrypoints
    version = _call_entrypoint_fn(root, config, ep.load())
  File "~\simulations\python\poppy\.eggs\setuptools_scm-6.0.1-py3.9.egg\setuptools_scm\__init__.py", line 46, in _call_entrypoint_fn
    return fn(root, config=config)
  File "~\simulations\python\poppy\.eggs\setuptools_scm-6.0.1-py3.9.egg\setuptools_scm\git.py", line 115, in parse
    require_command("git")
[... 36 lines cut ...]
        self.finalize_options()
      File "~\AppData\Local\Temp\pip-build-env-zpjfk00c\overlay\Lib\site-packages\setuptools\dist.py", line 743, in finalize_options
        ep(self)
      File "~\AppData\Local\Temp\pip-build-env-zpjfk00c\overlay\Lib\site-packages\setuptools\dist.py", line 750, in _finalize_setup_keywords
        ep.load()(self, ep.name, value)
      File "~\AppData\Local\Temp\pip-build-env-zpjfk00c\overlay\Lib\site-packages\setuptools_scm\integration.py", line 24, in version_keyword
        dist.metadata.version = _get_version(config)
      File "~\AppData\Local\Temp\pip-build-env-zpjfk00c\overlay\Lib\site-packages\setuptools_scm\__init__.py", line 173, in _get_version
        parsed_version = _do_parse(config)
      File "~\AppData\Local\Temp\pip-build-env-zpjfk00c\overlay\Lib\site-packages\setuptools_scm\__init__.py", line 127, in _do_parse
        version = _version_from_entrypoints(config) or _version_from_entrypoints(
      File "~\AppData\Local\Temp\pip-build-env-zpjfk00c\overlay\Lib\site-packages\setuptools_scm\__init__.py", line 65, in _version_from_entrypoints
        version = _call_entrypoint_fn(root, config, ep.load())
      File "~\AppData\Local\Temp\pip-build-env-zpjfk00c\overlay\Lib\site-packages\setuptools_scm\__init__.py", line 46, in _call_entrypoint_fn
        return fn(root, config=config)
      File "~\AppData\Local\Temp\pip-build-env-zpjfk00c\overlay\Lib\site-packages\setuptools_scm\git.py", line 115, in parse
        require_command("git")
      File "~\AppData\Local\Temp\pip-build-env-zpjfk00c\overlay\Lib\site-packages\setuptools_scm\utils.py", line 142, in require_command
[... cut ...]
WARNING: Discarding file:///~/simulations/python/poppy. Command errored out with exit status 1: '~\anaconda3\envs\testenv\python.exe' '~\anaconda3\envs\testenv\lib\site-packages\pip\_vendor\pep517\_in_process.py' prepare_metadata_for_build_wheel '~\AppData\Local\Temp\tmpfy1c1dqs' Check the logs for full command output.
ERROR: Command errored out with exit status 1: '~\anaconda3\envs\testenv\python.exe' '~\anaconda3\envs\testenv\lib\site-packages\pip\_vendor\pep517\_in_process.py' prepare_metadata_for_build_wheel '~\AppData\Local\Temp\tmpfy1c1dqs' Check the logs for full command output.
```
I hope this is helpfull else ask for more.

PR #416 (2021-03-31): Update installation docs
This PR clarifies the documentation on how to install poppy. Closes #410
closed 2021-03-31 by mperrin