spacetelescope/poppy #422: The fence problem in segmented apertures

Teusia, 2021-04-02:
Hello,

I have a problem of fence's poles and spacing: for some reason the CircularSegmentedDeformableMirror tells me that the diameter of my telescope is `(2n+1)*the_segment_size + (2n+1)*the_gap_size` (with n the number of rings ). When I am certain the diameter of my telescope is  `(2n+1)*the_segment_size + (2n)*the_gap_size`. Otherwise my fence is terminated by a non delimited spacing. The first version create unwanted zero-padding, also if you are relying on the header later for the pixel scale or the telescope diameter for the resolution, this one is wrong enough to cause weird bugs in the data reduction. (specially when you are working with ginormous gaps).

If we agree that the second method is the right method for calculating the telescope size, then I would like to offer the following fix:
In `poppy.optics.MultisegmentAperture` starting line 1262
```python
        self.pupil_diam = (self._segment_spacing) * (2 * self.rings + 1)
```
Replace by

```python
        self.pupil_diam = (segment_size) * (2 * self.rings + 1)+ gap * (2*rings)
```

Cheers,
Anne-Laure

mperrin, 2021-04-02:
Ah, this is a subtle point, and may benefit from different handling for circular and hexagonal apertures. First, there's a distinction between the pupil _array_ diameter and the pupil _clear aperture_ diameter. I agree with your formula for the clear aperture diameter. The `pupil_diam` variable is somewhat misnamed as it actually gets used to set the pupil array diameter. This is legacy old code but could be clarified. 

In the case of hexagonal segments we do intentionally want a little extra padding, because the circumscribing diameter is larger than the top-to-bottom distance across the aperture from one edge to the opposite side. This is not the case for circular apertures though, right. In your case you want the clear aperture to go right up against the edge of the array without zero padding. I think we can put in a fix for this which handles the two aperture shape subclasses differently, keeping the current behavior for hexagons but using your proposed fix for circles.

mperrin, 2021-04-02:
@Teusia please try adding your suggested replacement line in the function `MultiCircularAperture.__init__`, after the call to `super().__init__`. This is around line 1533. 

I think that should make things work as you desire for circular segments, without any change for the hexagonal case. 

If that works for you, please feel free to submit a pull request and we can merge in that fix. Thanks.

Teusia, 2021-04-06:
Hello,
@mperrin, that is a good point I did not think about that! okay no problem I'll do as you suggest!
I am submitting a pull request with your suggested change.

Thanks a lot!!!

closed 2021-04-06 by Teusia