spacetelescope/poppy #430: FITSOpticalElements in image plane: unit problem

Teusia, 2021-04-27:
Hello,

I am trying to use the `FITSOpticalElement` to introduce a pyramid wavefront sensor in a poppy optical system. Only I cannot get my final image for what appear to be a unit problem:
```python
Traceback (most recent call last):

  File "~\anaconda3\envs\poppystable\lib\site-packages\astropy\units\quantity_helper\helpers.py", line 32, in get_converter
    scale = from_unit._to(to_unit)

  File "~\anaconda3\envs\poppystable\lib\site-packages\astropy\units\core.py", line 950, in _to
    raise UnitConversionError(

UnitConversionError: 'Unit("m / pix")' is not a scaled version of 'Unit("arcsec / pix")'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "~\anaconda3\envs\poppystable\lib\site-packages\astropy\units\quantity_helper\helpers.py", line 77, in get_converters_and_unit
    converters[changeable] = get_converter(unit2, unit1)

  File "~\anaconda3\envs\poppystable\lib\site-packages\astropy\units\quantity_helper\helpers.py", line 34, in get_converter
    return from_unit._apply_equivalencies(

  File "~\anaconda3\envs\poppystable\lib\site-packages\astropy\units\core.py", line 886, in _apply_equivalencies
    raise UnitConversionError(

UnitConversionError: 'm / pix' and 'arcsec / pix' are not convertible

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "~\simulations\python\PhasingSimulations\mypackage\telescopePyramid.py", line 86, in <module>
    psf = osys.calc_psf(wavelength = wavelength)#the error show at this line
#... I shorten this second error message for readability
  File "~\anaconda3\envs\poppystable\lib\site-packages\astropy\units\quantity_helper\helpers.py", line 79, in get_converters_and_unit
    raise UnitConversionError(

UnitConversionError: Can only apply 'subtract' function to quantities with compatible dimensions
```
I think solving the two first problem fixes the last one, so I'll concentrate on the first problem:
The pixel scale of this `FITSOpticsElement` should be in arcsec/pix, if I understand properly, because it will be an image plan. No matter what I do the pixel scale of my `FITSOpticsElement` is in unit of m/pix. 
As recommended by the documentation, I have in my header a `PIXUNIT` and tried to make it arcsecond, arcsecondes, arcsec/pix and arcsec. None of them changes the unit of the pixel scale. 
I also tried to force the plane type to image (planetype = 2) that does not help either.
I cannot find in the source code where the `PIXUNIT` is being read so I do not know what else to do.

Since the problem is most likely that I am doing something wrong, here is an example code I am working with to try this out:
```python
# -*- coding: utf-8 -*-
import numpy as np
import matplotlib.pyplot as plt
from astropy.io import fits
import poppy

wavelength = 0.850*10**-6 
#my telescope parameters
nring = 1
segmentDiameter = 8.365 
gap = 0.3
secondarySize = 3.22
puprotation = 15

#some further calculations 
# segSepa = segmentDiameter+gap
# nseg = (nring**2*3)+3*nring + 1
telescopeDiameter = segmentDiameter*3+2*gap

npixel = 512

#%% create the fits optic file
#do not forget to erase the last created fits optics when rerunning this section
gmtpixelScale = 6.89*10**-6 #m/pix This is 1 lambda/D equivalent on GMT plate scale

pyramidSlope = 30 #degree
pyramidOPD = np.zeros((npixel,npixel))

#first we need a coordinate system:
X,Y = np.meshgrid(np.arange(-npixel/2,npixel/2),np.arange(-npixel/2,npixel/2 ))
X+=0.5
Y+=0.5
X *= gmtpixelScale
Y*=gmtpixelScale

#Then we need to create the slope based on the coordinates
tmpquadrant = (-X-Y)/2*np.tan(pyramidSlope*np.pi/180)
#and we add it to the appropriate corner of the OPD
pyramidOPD[(X>0) * (Y>0)] = tmpquadrant[(X>0) * (Y>0)]

#repeat
tmpquadrant = (X+Y)/2*np.tan(pyramidSlope*np.pi/180)
pyramidOPD[(X<0) * (Y<0)] = tmpquadrant[(X<0) * (Y<0)]

tmpquadrant = (X-Y)/2*np.tan(pyramidSlope*np.pi/180)
pyramidOPD[(X<0) * (Y>0)] = tmpquadrant[(X<0) * (Y>0)]

tmpquadrant = (-X+Y)/2*np.tan(pyramidSlope*np.pi/180)
pyramidOPD[(X>0) * (Y<0)] = tmpquadrant[(X>0) * (Y<0)]

# plt.imshow(pyramidOPD,origin = 'lower')

#now we can save the result in a poppy compatible format:
hdu = fits.PrimaryHDU(pyramidOPD)
hdu.header['WAVELEN'] = wavelength
hdu.header['DIAM'] = telescopeDiameter
hdu.header['DIFFLMT'] = wavelength/telescopeDiameter
hdu.header['OVERSAMP'] = 1
hdu.header['DET_SAMP'] = 1
hdu.header['PIXELSCL'] = telescopeDiameter/npixel 
hdu.header['PIXUNIT'] = 'arcsecond '
hdu.header['BUNIT'] = 'nanometers'

hdul = fits.HDUList(hdu)
hdul.writeto('pyramidOPD.fits') 

#%% Now I do a poppy calculation of the optical system using this script
pixscale = telescopeDiameter/npixel

#here I define the surfaces
pupildms = poppy.dms.CircularSegmentedDeformableMirror(name = 'segCorrector', rings = nring, segment_radius=segmentDiameter/2,gap = gap,rotation = puprotation)
pyrFoc = poppy.FITSOpticalElement(opd = 'pyramidOPD.fits',pixelscale = "PIXELSCL",planetype=2 )
print(pyrFoc.pixelscale)
det = poppy.Detector(name='detector', pixelscale=telescopeDiameter/npixel, fov_pixels=npixel)

#Now The poppy optical system is being built
osys = osys = poppy.OpticalSystem()
osys.add_pupil(pupildms)
osys.add_image()
osys.add_image(pyrFoc)#the problem disapear if i comment this line hence I know my fits optics is creating the problem
osys.add_pupil(det)
#Calculate the image/pupil plan 
psf = osys.calc_psf(wavelength = wavelength)#the error show at this line
```
How can I make this code work, please? Or is this just not going to work? (because this is the wrong way to do it?)
Cheers,
Anne-Laure

mperrin, 2021-04-29:
Hi @Teusia, 

I think here the fix is simple but not obvious: Replace `planetype=2` with `planetype=poppy.poppy_core.PlaneType.image`.  The python enum class checks not just the numeric value but also the class type, so you have to set that parameter with a PlaneType class instance. With that change, your example code runs without any error. 

I hope this helps!

Teusia, 2021-04-30:
Hello @mperrin,
Yes that did the trick the error is no longer there! thank you!
Now I want to understand why it seems the pyramid has no effect on the image.
Any insight maybe?

Thank you!

closed 2021-05-07 by Teusia