spacetelescope/poppy #452: StatisticalPSDWFE divide by zero

philhinzatucsc, 2021-08-04:
I have a newbie question/ bug.  I have just started using poppy to estimate wavefront error effects on the PSF for Keck/SCALES.  To do so, I have a created a FresnelOptical System model and then use StatisticalPSDWFE to create an error for one of the optics.  This results in a warning:

 . . ./poppy/wfe.py:349: RuntimeWarning: divide by zero encountered in power
  psd = np.power(rho, -self.index)   # generate power-law PSD

I see why it's complaining. r^-3 is infinite when r=0 (center of the array).  So I went to  wfe.py and added a really crude hack:
    rho[rho == 0] = 0.00001   #get rid of infinity.  

Now this works fine and I can happily continue modeling.  Is this a real bug, or am I missing something about how to properly use StatisticalPSDWFE?

mperrin, 2021-08-04:
@jlumbres could you please take a look at this issue and provide your take on this divide by zero issue with StatisticalPSDWFE? Thanks much..

jlumbres, 2021-08-05:
Oh, hi @philhinzatucsc! 

This issue happened with me when I wrote PowerSpectrumWFE. I did a similar solution, too. I overwrote the one pixel where spatial frequency k=0 to be 1*dk (spatial frequency spacing), did my math, then I overwrote the power at that same pixel location to be 0 after the math was done. I set the power to 0 because it was an arbitrary choice on my end. You can see my code [here](https://github.com/spacetelescope/poppy/blob/c189bceab789ee8ca22f6ffa54272bcfc095a356/poppy/wfe.py#L531).

philhinzatucsc, 2021-08-05:
Ok, thanks for the confirmation.  I am new enough to github and poppy that I am not sure what the next step is.  For my own work, my internal hack works fine, but presumably I or you should be fixing this for others?