spacetelescope/poppy #475: GaussianAperture ignores pupil_diam

retroexpress, 2021-12-07:
It seems the pupil aperture is always 3*FWHM when I generate a Gaussian aperture, even when specified otherwise. Here are the plots for pupil_diam=0.01 and 1:

```
npix = 1024
wf = poppy.FresnelWavefront(0.1*u.m,wavelength=1550e-9,npix=npix,oversample=4)
wf *= poppy.GaussianAperture(w=2.28*15/2*u.mm,pupil_diam=1)
```

[image]
[image]

This is an awesome software package BTW.

mperrin, 2021-12-16:
Hi @retroexpress - I just wanted to say, I/we did see this issue report. Apologies but we won't be able to look into it in any detail for some time (weeks, possibly months) - all our efforts right now are focused on supporting the immanent launch and commissioning of the Webb Space Telescope. 

That said I am not quite sure what exactly is the issue you're reporting. In the example code given, the wavefront size is set when it's created: 
```
wf = poppy.FresnelWavefront(0.1*u.m,wavelength=1550e-9,npix=npix,oversample=4)
```
That sets the array size and extent, independent of any optics. 

To clarify, the `pupil_diam` parameter of an optic such as the GaussianAperture is only relevant in cases where there isn't a sampling defined by some Wavefront. For instance, if you try to display that aperture on its own via the `.display()` method, it needs to pick some pupil size to sample and display the optic on. But in the case you show here, the sampling is set first on the Wavefront object.

retroexpress, 2021-12-21:
Hi Marshall, thanks for the reply. I understand you have some pretty big fish to fry for the foreseeable future so I really appreciate you taking the time. The issue I was reporting was that size of the aperture of the gaussian always seems fixed at 3*FWHM, regardless of what I specify. Is pupil_diam not what is specifying the size of the circle in the phase plots?