spacetelescope/poppy #492: Poppy Utility Functions Issues

MTLeist255, 2022-02-09:
Greetings, I am attempting to use the following functions 'display_ee, radial_profile, display_profiles' to measure and display the FWHM, radial profile, and encircled energies of a few model inputs and have run into the following issues I was hoping to get some help in resolving.

1) **_display_ee_**: Not displaying radial percentages and values, unable to add a title
[image]
Here is the line of code I am calling to create this plot:
_display_ee(residual, overplot = False, mark_levels = True)_

2) **_radial_profiles_**: not populating at all
[image]
Here is the line of code I am calling to create this plot:
_radial_profile(residual, ee = False, stddev = True)_

3) **_display_profiles_**: not measuring(displaying?) correct FWHM measurements (I am measuring a 2D-Gaussian I created with a known FWHM of 2), not displaying radial percentages and values for the encircled energy, and not including title
[image]
Here is the line of code I am calling to create this plot:
_display_profiles(residual, overplot = False, center = position2, title='FWHM Test')_

I am curious if I am attempting to measure an incorrectly formatted FITS file and if this is the reason I am receiving my errors. I've searched for online examples using these utility functions and documentation to help address these issues I am having. I've found the source code here https://www.stsci.edu/~mperrin/software/poppy_prerelease/_modules/poppy/utils.html but not a lot of examples. Is there a repository somewhere with examples on the proper ways to format your FITS file to be read by these utilities, with different examples demonstrating their capabilities?

Best Regards,
Mason

mperrin, 2022-02-16:
Hi Mason, Thanks for the issue report. Apologies for not having time to respond right now - right now all of us who develop & maintain poppy are excitingly busy commissioning a 6.5 meter space telescope. Thanks for your understanding and patience!

For examples of how to format FITS files, you can look at any of the output PSF files as produced by poppy itself.  Looking at your plots above it seems to me you probably have not normalized your PSFs. The utility functions in general expect the PSF to be in units normalized to total intensity = 1.0. That may be part of the issue you're having.

MTLeist255, 2022-03-08:
Hi mperrin, I was wondering if there is an updated version of this code available? I've normalized my PSFs to a total intensity = 1.0, but am having trouble returning accurate FWHM measurements in the presence of noise. This tool works well if measuring the radial profile, encircled energy and FWHM of a WebbPSF generated image, but fails when trying to measuring a MIRISim simulated image. I was wondering what advice you could give to help solve this problem?
Regards, Mason Leist