spacetelescope/poppy #502: factorial() won't accept floats in python >=3.10

tepickering, 2022-04-01:
i added python 3.10 to my test matrices and i get failures due to `poppy`. the specific error is:
```
        if _is_odd(n - m):
            return 0
        else:
            for k in range(int((n - m) / 2) + 1):
                coef = ((-1) ** k * factorial(n - k) /
>                       (factorial(k) * factorial((n + m) / 2. - k) * factorial((n - m) / 2. - k)))
E               TypeError: 'float' object cannot be interpreted as an integer

../py310-alldeps/lib/python3.10/site-packages/poppy/zernike.py:174: TypeError
```
`factorial()` won't accept floats as of python 3.10: https://docs.python.org/3/library/math.html#math.factorial.

mperrin, 2022-04-01:
Hey Tim, yes this is fixed in the current dev version, so just grab the current source from here on GitHub and you should be good to go. Was fixed in a recent PR, though we haven't gotten around to doing a release yet (It's been a little busy this month, in a very good way :-)  But since we've gotten a couple messages on this issue this week alone, clearly we probably ought to get one out before too long!

Cheers, -M

mperrin, 2022-04-01:
@shanosborne See above, and I got an email yesterday on this same issue from @goudfroo too. We may want to fit in a poppy release sometime soon, for compatibility with python 3.10's factorial function changes.

tepickering, 2022-04-01:
sounds good. i'm using 3.9 in production so not a huge deal. i also have the 3.10 tests set to allow failure.

shanosborne, 2022-04-04:
@mperrin Good idea, I'm happy to work on a poppy release whenever we have time. We can drop 3.7 support and add 3.10  in the CI to go along with that release. I'll work on PRs for that for webbpsf and poppy

brantr, 2022-04-26:
Just re-upping that this is still a blocking issue for Python 3.10 as poppy v1.0.2 is still current via pip, prevents users from running the WebbPSF demo at https://webbpsf.readthedocs.io/en/stable/usage.html

jrhastro, 2022-06-17:
@shanosborne @mperrin Will email off-thread, but I would like to re-up this issue as well.  This is currently a blocking issue for an upcoming release of Roman STIPS.

ojustino, 2022-06-30:
I'm closing this issue because the fix should be in today's new release of `poppy`, version 1.0.3.

PR #510 (2022-06-30): Release 1.0.3
Crystallizes the latest changes to the development branch.

(Via #493, would close #502.)
closed 2022-06-30 by ojustino