spacetelescope/poppy #526: Simple optical system aliasing, known behaviour or issue?

Teusia, 2022-09-12:
Dear all,

I ran into a behaviour I would qualify as weird, but maybe I am just using poppy the wrong way. 
If I use a large field of view and a small wavelength, it looks like there is some fun Fourier transform aliasing happening. 

[image]

This is the code I used to produce the above PSF. 
```python
import poppy

pup = poppy.CircularAperture(radius = 39)

osys = poppy.OpticalSystem()

osys.add_pupil(pup)
osys.add_detector(pixelscale=0.001, fov_arcsec=2.0)

psf = osys.calc_psf(0.3e-6)

poppy.display_psf(psf)
```
I am using this build '1.0.4.dev5+g08d1eaa.d20220808' of poppy which according to git is the latest in the branch stable. 

Is this an expected behaviour? or I did something wrong?
Best Regards,

Rikyf3, 2025-01-11:
What about that I am seeing 9 PSFs instead of 1? Is it a real world phenomena? If not how can I fix that?

obi-wan76, 2025-01-18:
It is a numerical artifact due to aliasing and the implicit repetition of the pupil entrance aperture in the discrete Fourier transform. If you, for instance, **increase** the wavelength by a factor of 4 or **decrease** the pupil by a factor of 4, then you don't get repetition
e.g.

```
import poppy

pup = poppy.CircularAperture(radius = 10) # as an example decrease by fator of 4 from 39 to 10

osys = poppy.OpticalSystem()

osys.add_pupil(pup)
osys.add_detector(pixelscale=0.001, fov_arcsec=2.0)

psf = osys.calc_psf(0.3e-6)  # or, alternatively, increase by factor of 4, from 0.3 to 1.2

poppy.display_psf(psf)
```

[image]