spacetelescope/poppy #544: np.float deprecated as of numpy v1.24.0

adambolton, 2023-02-08:
`dtype=np.float` as on lines 66-67 of `geometry.py` is deprecated and breaks under recent versions of numpy.

Replacement with `np.float32` or `np.float64` should solve.

I could attempt the fix here and elsewhere if given guidance as to whether to use 32 or 64.

douglase, 2023-02-09:
Since elsewhere in poppy `complex128` is needed, https://github.com/spacetelescope/poppy/blob/4e25b8ec6e886b0514215fadd215d423a1c76ae9/poppy/optics.py#L127, and the python `float` was [originally](https://docs.python.org/2/library/stdtypes.html#numeric-types-int-float-long-complex) a 64 bit double I think this would be safest to become`dtype=np.float64`.

douglase, 2023-02-09:
alternatively but with the same outcome, the default NumPy datatype is `np.float_` which is an alias to double https://numpy.org/devdocs/reference/arrays.scalars.html#numpy.float_

mperrin, 2023-02-09:
I like the suggestion of `np.float_` since that’s the default data type. 

@adambolton if you could put together a fix for this and pull request it in, that would be much appreciated! Thanks for the heads up about this change in recent numpy.

mperrin, 2023-02-09:
@BradleySappington and @obi-wan76 FYI, see this thread. This is a good example of a simple software maintenance/dependencies issue with poppy (thus also with WebbPSF) which has slipped through the cracks since Shannon’s departure. 
In fact I see there’s a dependabot PR which fails because of this numpy change, https://github.com/spacetelescope/poppy/pull/538, which could have brought this to our attention a while ago. Except nobody’s been looking at the dependabot PRs lately after Shannon left. Glad you will be able to start lending a hand for all this.

adambolton, 2023-02-09:
I'll take a run at this tomorrow (Fri 10 Feb 2023). Cheers!

BradleySappington, 2023-02-09:
Thanks for the link @mperrin, will start keeping an eye on the repository

mperrin, 2023-03-03:
Fixed via #545

closed 2023-03-03 by mperrin