spacetelescope/poppy #553: Propagation through turbulent atmosphere notebook is broken

BenoitSchillings, 2023-03-21:
There seemed to be an issue with the unit system for the KolmogorovWFE function. the example will not run.

BenoitSchillings, 2023-03-30:
Here is the error log :

---------------------------------------------------------------------------
UnitConversionError                       Traceback (most recent call last)
File ~/.local/lib/python3.10/site-packages/astropy/units/quantity_helper/helpers.py:141, in helper_dimensionless_to_dimensionless(f, unit)
    140 try:
--> 141     return ([get_converter(unit, dimensionless_unscaled)], dimensionless_unscaled)
    142 except UnitsError:

File ~/.local/lib/python3.10/site-packages/astropy/units/quantity_helper/helpers.py:35, in get_converter(from_unit, to_unit)
     33 """Like Unit._get_converter, except returns None if no scaling is needed,
     34 i.e., if the inferred scale is unity."""
---> 35 converter = from_unit._get_converter(to_unit)
     36 return None if converter is unit_scale_converter else converter

File ~/.local/lib/python3.10/site-packages/astropy/units/core.py:1094, in UnitBase._get_converter(self, other, equivalencies)
   1092                 return lambda v: b(converter(v))
-> 1094 raise exc

File ~/.local/lib/python3.10/site-packages/astropy/units/core.py:1077, in UnitBase._get_converter(self, other, equivalencies)
   1076 try:
-> 1077     return self._apply_equivalencies(
   1078         self, other, self._normalize_equivalencies(equivalencies)
   1079     )
   1080 except UnitsError as exc:
   1081     # Last hope: maybe other knows how to do it?
   1082     # We assume the equivalencies have the unit itself as first item.
   1083     # TODO: maybe better for other to have a `_back_converter` method?

File ~/.local/lib/python3.10/site-packages/astropy/units/core.py:1054, in UnitBase._apply_equivalencies(self, unit, other, equivalencies)
   1052 other_str = get_err_str(other)
-> 1054 raise UnitConversionError(f"{unit_str} and {other_str} are not convertible")

UnitConversionError: 'm' (length) and '' (dimensionless) are not convertible

During handling of the above exception, another exception occurred:

UnitTypeError                             Traceback (most recent call last)
Cell In[6], line 11
      8         wf.propagate_fresnel(dz)
      9         phase_screen = poppy.KolmogorovWFE(Cn2=Cn2, dz=dz)
---> 11     wf *= phase_screen
     13 fig = plt.figure(figsize=(12,5))
     14 fig.clf()

File ~/.local/lib/python3.10/site-packages/poppy/fresnel.py:915, in FresnelWavefront.__imul__(self, optic)
    912     return self
    913 else:
    914     # Otherwise fall back to the parent class
--> 915     return super(FresnelWavefront, self).__imul__(optic)

File ~/.local/lib/python3.10/site-packages/poppy/poppy_core.py:181, in BaseWavefront.__imul__(self, optic)
    178     self.location = 'at ' + optic.name
    179     return self
--> 181 phasor = optic.get_phasor(self)
    183 if not np.isscalar(phasor) and phasor.size > 1:
    184     assert self.wavefront.shape == phasor.shape, "Phasor shape {} does not match wavefront shape {}".format(
    185         phasor.shape, self.wavefront.shape)

File ~/.local/lib/python3.10/site-packages/poppy/optics.py:143, in AnalyticOpticalElement.get_phasor(self, wave)
    140         return np.asarray(result, _complex())
    142 else:
--> 143     return self.get_transmission(wave) * np.exp(1.j * self.get_opd(wave) * scale)

File ~/.local/lib/python3.10/site-packages/astropy/units/quantity.py:643, in Quantity.__array_ufunc__(self, function, method, *inputs, **kwargs)
    621 """Wrap numpy ufuncs, taking care of units.
    622 
    623 Parameters
   (...)
    637     Results of the ufunc, with the unit set properly.
    638 """
    639 # Determine required conversion functions -- to bring the unit of the
    640 # input to that expected (e.g., radian for np.sin), or to get
    641 # consistent units between two inputs (e.g., in np.add) --
    642 # and the unit of the result (or tuple of units for nout > 1).
--> 643 converters, unit = converters_and_unit(function, method, *inputs)
    645 out = kwargs.get("out", None)
    646 # Avoid loop back by turning any Quantity output into array views.

File ~/.local/lib/python3.10/site-packages/astropy/units/quantity_helper/converters.py:182, in converters_and_unit(function, method, *args)
    179 units = [getattr(arg, "unit", None) for arg in args]
    181 # Determine possible conversion functions, and the result unit.
--> 182 converters, result_unit = ufunc_helper(function, *units)
    184 if any(converter is False for converter in converters):
    185     # for multi-argument ufuncs with a quantity and a non-quantity,
    186     # the quantity normally needs to be dimensionless, *except*
   (...)
    189     # can just have the unit of the quantity
    190     # (this allows, e.g., `q > 0.` independent of unit)
    191     try:
    192         # Don't fold this loop in the test above: this rare case
    193         # should not make the common case slower.

File ~/.local/lib/python3.10/site-packages/astropy/units/quantity_helper/helpers.py:143, in helper_dimensionless_to_dimensionless(f, unit)
    141     return ([get_converter(unit, dimensionless_unscaled)], dimensionless_unscaled)
    142 except UnitsError:
--> 143     raise UnitTypeError(
    144         f"Can only apply '{f.__name__}' function to dimensionless quantities"
    145     )

UnitTypeError: Can only apply 'exp' function to dimensionless quantities

BenoitSchillings, 2023-04-25:
Any chance for a solution ?

mperrin, 2023-05-31:
The fix for this was released, version 1.1.0, so I'm marking this closed.