spacetelescope/poppy #560: New Poppy Installation Broken?

Kokomodo, 2023-04-14:
Hi,

I'm new to POPPY and have basic to intermediate python skills.  I installed the poppy package with the dependencies as described in requirements.txt (as well as synphot_refactor) but get the following error when running the first (most simple) example:

AttributeError: module 'numpy' has no attribute 'float'.

`np.float` was a deprecated alias for the builtin `float`. To avoid this error in existing code, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.

The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:

    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations

Has numpy been updated since the last poppy release?  The versions all match the requirements.txt file.

This is my code:

  1  import poppy

  2 import scipy

  3 import numpy

  4 import matplotlib

  5 import synphot_refactor

  6 poppy.__version__

  7 numpy.__version__

  8 print("Hello World.")

  9 osys = poppy.OpticalSystem()

 10 osys.add_pupil( poppy.CircularAperture(radius=3))    # pupil radius in meters

 11 osys.add_detector(pixelscale=0.010, fov_arcsec=5.0)  # image plane coordinates in arcseconds

 12 

 13 psf = osys.calc_psf(2e-6)                            # wavelength in meters

 14 poppy.display_psf(psf, title='The Airy Function')

~

Kokomodo, 2023-04-14:
Obviously lines 6 and 7 are incorrect.  I'm not sure why the text edit lost the underscores, and yes they should be inside print calls.  For some reason I cannot highlight the code in the issue post to format it properly.

douglase, 2023-04-14:
this looks like https://github.com/spacetelescope/poppy/issues/544 which was fixed in https://github.com/spacetelescope/poppy/pull/545, until the next release you might try try cloning the develop branch and installing that (option 2, https://poppy-optics.readthedocs.io/en/latest/installation.html):

``` 
git clone https://github.com/spacetelescope/poppy.git
cd poppy
pip install -e .
```

Kokomodo, 2023-04-15:
Actually, I'm pretty sure I cloned the repository the first time I installed it rather than using pip.  Given that I never made it past the first example, I may blow out the first install and try again.  Maybe I hosed something up in the environment?

closed 2023-04-15 by Kokomodo