spacetelescope/poppy #572: Unexpected modification of the FresnelWavefront matrix size after applying rotation

SteveOstile, 2023-06-05:
Dear Poppy team,

Thanks for nice work on Poppy !

I've found an unexpected modification of the FresnelWavefront matrix size after applying rotation using the following code :
```python

import matplotlib.pyplot as plt
import numpy as np
import poppy
import astropy.units as u

w0 = 0.1 # in meter
lamb = 532e-9 # in meter

wf0 = poppy.FresnelWavefront(beam_radius=w0*u.m, wavelength=lamb)
ap = poppy.CircularAperture(radius=w0)
wf0 *= ap

#generate tilt to better see the following rotation
wf0.tilt(Xangle=10.0)

#Display before rotation
plt.figure(figsize=(12,5))
wf0.display('both',colorbar=True, showpadding=True)
plt.suptitle("Before Rotation", fontsize=18)

#Copy wavefront and apply 90° rotation
wf1 = wf0
print('Before rotation:',np.shape(wf1))
wf1.rotate(90.0)
print('After rotation:',np.shape(wf1))

#Display after rotation
plt.figure(figsize=(12,5))
wf1.display('both',colorbar=True, showpadding=True)
plt.suptitle("After rotation", fontsize=18)

``` 
The 2 print outputs are:

Before rotation: (2048, 2048)
After rotation: (1024, 1024)

Displaying amplitude and phase before rotation:
[image]

Displaying amplitude and phase before rotation:
[image]

As you can see, the matrix size goes from 2048 x 2048 to 1024x1024 and the width of the beam is doubled.

Thanks for looking at this issue.

mperrin, 2023-06-05:
Oh and @SteveOstile: welcome! Thanks for reporting this. And I see this looks like your first comment on a GitHub issue so let me extend an extra welcome in that case :-)

mperrin, 2023-06-05:
Fix added in PR #573.  @SteveOstile, let me know if you're comfortable enough with git/gitub to perhaps check out and test that code? For me, running with the modified version in the PR, your example code above runs with no surprises.

SteveOstile, 2023-06-06:
Hi @mperrin,

Thanks for reacting so quickly to my comment! I understand the issue now. 
Well, I am so new on github that I am not very comfortable in reviewing your modification. I have to look how I can pip install your modification...

closed 2023-06-07 by mperrin

mperrin, 2023-06-07:
Hi @SteveOstile we've merged the fix into the `develop` branch here. I believe you should be able to install from GitHub with something like this:
```
pip install git+https://github.com/spacetelescope/poppy.git@develop#egg=poppy
```

SteveOstile, 2023-06-08:
@mperrin Excellent ! It works perfectly !

commit 53286cee (2023-06-07): Merge pull request #573 from mperrin/fix_572

Avoid un-padding Fresnel wavefronts when rotating (fixes #572)