spacetelescope/poppy #638: Thermal blooming calculations doesn't give plausibleresults for other mediums such as water

CentralStar, 2024-09-05:
I used the thermal blooming routine for water but I can't seem to get it  work. I appreciate any help.  Water pipe is 1 m long and laser power is about 100 W. I tried other power levels and convective speeds but didn't get to see the effect. I can see thermal blooming effect in the lab for this type of scenario.

mperrin, 2024-09-20:
Hi @CentralStar and thanks for your patience awaiting a response on this.  Can you provide example code that reproduces the issue that you are encountering, please?  Example code and outputs would aid in investigating. 

While I would like to help, I  don't have much/any direct expertise with the thermal blooming implementation. This was contributed open source by @DaPhil, who may or may not be able to advise on this. And like I said, example code showing what you're doing and the outputs of the code would be appreciated. Cheers.

DaPhil, 2024-09-20:
Dear @CentralStar, as @mperrin suggested, please provide the exact input you have used and the results produced. Please also state, why you think the results are not plausible and what you expected to see.

I developed the code about 8 years ago. I since left the academic realm and have no longer access to either computing power nor the papers used to model the implementation. You should check the relevant papers, if the assumptions made for the implemented equations also apply for water. You can find the relevant papers in the wfe.py file in the ThermalBloomingWFE class. If I remember correctly, the papers from Fleck and Smith were most relevant. I also remember that for the non-isobaric approach, one should take care about the numerical grid since you need to start at some edge of the grid to calcuate the density variations.

Cheers
Phil