spacetelescope/poppy #645: Schedule for next release?

braingram, 2024-12-06:
Warnings are showing up in romancal testing (which includes poppy indirectly through webbpsf). Here's one example that shows up as a DeprecationWarning in python 3.11 (and a SyntaxWarning in python 3.12):
```
poppy/zernike.py:103: DeprecationWarning: invalid escape sequence '\s'
    return "\sqrt{%d}* ( %s ) * \\cos(%d \\theta)" % (2 * (n + 1), outstr, m)
```
It appears this was fixed in https://github.com/spacetelescope/poppy/commit/d2be71e7feee669e685ff9f1c71982482acd045c which has not yet been released. Is it possible to make a release to publish these changes?

mperrin, 2024-12-06:
@braingram This is a good question. To be honest we hadn't yet made a specific plan for this but I agree it's timely due to these syntax warnings, and a few other small tweaks. We're planning a webbpsf release in the immediately future (some updates needed to accompany the most recent SOC release which just happened this week), and it works well to do a release of this now too. @BradleySappington agreed to help and we'll try to roll this out in the very near future (this coming week, perhaps?)

BradleySappington, 2025-01-14:
@braingram 1.1.2 has been released. Closing this issue.

closed 2025-01-14 by BradleySappington